shivasiddharth/Stremio-RaspberryPi #1: Subtitles Addon

adfcorreia, 2020-04-13:
After installing the software on rpi4, the subtitles add-on does not work. The following message is displayed: "Error loading subtitles, no addon supplies this method/argument"

atomms, 2020-05-06:
try changing to last stremio client:
http://app.strem.io/shell-v4.4/#/
in your /home/${USER}/Stremio-RaspberryPi/src/client-launcher.sh

atomms, 2020-05-06:
by the way i still have problems with a slow performance -stremio client freezes again and again- until i open the stream with a different client such as omxplayer, but this omxplayer does not load subtitles as the default client does

stuck in the middle of this :/

shivasiddharth, 2020-05-11:
Can you see if VLC works ? We have circumvented few packages to get Stremio running. So obviously you will not have the performance of the desktop version.

atomms, 2020-05-22:
VLC does not start when i launch it from within server.js as with default options:

title: "VLC",
            args: ['--no-video-title-show'],
            subArg: '--sub-file=',
            timeArg: '--start-time=',
            playArg: '',
            darwin: {
                path: ["/Applications/VLC.app/Contents/MacOS/VLC"]
            },
            linux: {
                path: ['/usr/bin/vlc', '/usr/local/bin/vlc']
            },

VLC is working though from Desktop...

do you think it is about wrong vlc arguments?

eldroan, 2020-10-25:
Having the same issues, have you got it to work?

atomms, 2020-12-19:
not really, still needing VLC to play but can't see or retrieve any controls in VLC (play/pause/subtitles/...) after I right click the mouse to switch to VLC from the default player which freezes very often while playing streams.. 

any ideas here?

shivasiddharth, 2021-07-04:
Issues with VLC Have been fixed in the last update. It supports controls and subtitles. Forgot to update here.

PR #26 (2021-07-07): VLC Issues Fixes
Fixes #25  Fixes #1 
1. Services are run as user and not as root user.
2. VLC and other media player controls work even for services.
3. A minor delay has been introduced to the start of client service.
closed 2021-07-04 by shivasiddharth