shivasiddharth/Stremio-RaspberryPi #2: Auto start on boot not working

guardabrazo, 2020-05-10:
I've followed the steps mentioned to auto start Stremio on boot but I'm getting several errors:

```
/lib/systemd/system/stremio-client.service:3: Failed to add dependency on network-online.target
/lib/systemd/system/stremio-client.service:3: Failed to add dependency on stremio-server
/lib/systemd/system/stremio-client.service:4: Failed to add dependency on network-online.target
/lib/systemd/system/stremio-client.service:4: Failed to add dependency on network-online.target
```

shivasiddharth, 2020-05-11:
IS the server and client running on same Pi ? Can you post the outputs of 
```
sudo systemctl status stremio-server.service   
sudo systemctl status stremio-client.service   
```

guardabrazo, 2020-05-11:
Yeah, both server and client on the same PI. The server loads just fine but the client doesn't. If I run the client manually it works just fine.

Here are the outputs that you requested:
```
● stremio-server.service - Stremio Media Server
   Loaded: loaded (/lib/systemd/system/stremio-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-05-11 21:01:54 CEST; 5min ago
 Main PID: 432 (server-launcher)
    Tasks: 12 (limit: 2200)
   Memory: 54.9M
   CGroup: /system.slice/stremio-server.service
           ├─432 /bin/bash /home/pi/Stremio-RaspberryPi/src/server-launcher.sh
           └─435 node /home/pi/Stremio-RaspberryPi/stremio-shell/server.js

May 11 21:01:54 raspberrypi systemd[1]: Started Stremio Media Server.
May 11 21:02:03 raspberrypi server-launcher.sh[432]: hls executables located ->  { ffmpeg: '/usr/bin/ffmpeg', ffsplit: n
May 11 21:02:04 raspberrypi server-launcher.sh[432]: Using app path -> /home/pi/.stremio-server
May 11 21:02:04 raspberrypi server-launcher.sh[432]: Enabling casting...
May 11 21:02:04 raspberrypi server-launcher.sh[432]: Discovery of new external device "omx" - OMX
May 11 21:02:04 raspberrypi server-launcher.sh[432]: Discovery of new external device "vlc" - VLC
May 11 21:02:04 raspberrypi server-launcher.sh[432]: Discovery of new external device "mpv" - MPV
May 11 21:02:05 raspberrypi server-launcher.sh[432]: EngineFS server started at http://127.0.0.1:11470
May 11 21:02:05 raspberrypi server-launcher.sh[432]: EngineFS HTTPS endpoint at https://local.strem.io:12470
lines 1-19/19 (END)
```
```
● stremio-client.service - Stremio Media Client
   Loaded: loaded (/lib/systemd/system/stremio-client.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-05-11 21:02:12 CEST; 6min ago
  Process: 1010 ExecStart=/home/pi/Stremio-RaspberryPi/src/client-launcher.sh (code=exited, status=1/FAILURE)
 Main PID: 1010 (code=exited, status=1/FAILURE)

May 11 21:02:12 raspberrypi systemd[1]: stremio-client.service: Failed with result 'exit-code'.
May 11 21:02:12 raspberrypi systemd[1]: stremio-client.service: Service RestartSec=100ms expired, scheduling restart.
May 11 21:02:12 raspberrypi systemd[1]: stremio-client.service: Scheduled restart job, restart counter is at 7.
May 11 21:02:12 raspberrypi systemd[1]: Stopped Stremio Media Client.
May 11 21:02:12 raspberrypi systemd[1]: stremio-client.service: Start request repeated too quickly.
May 11 21:02:12 raspberrypi systemd[1]: stremio-client.service: Failed with result 'exit-code'.
May 11 21:02:12 raspberrypi systemd[1]: Failed to start Stremio Media Client.
```

PR #6 (2020-09-25): Update
1. Fixed auto start issue. Fixes #2 
2. Fixed the addon-issue. All the addons are available now.
closed 2020-09-25 by shivasiddharth