shivasiddharth/Stremio-RaspberryPi #8: Failed to fetch

IEYTradin, 2020-11-17:
Problem when trying to stream. All add-ons give "failed to fetch". Any suggestions?

koStelCk, 2020-11-24:
I have the same problem after I get it to autostart

IEYTradin, 2020-11-24:
Do you only have the problem when auto starting? I didn’t try that

IEYTradin, 2020-11-24:
Do you only have the problem when auto starting? Because I didn’t try it without auto starting

koStelCk, 2020-11-24:
when I try it without auto starting it works.. to test it out try
sudo systemctl disable stremio-server.service   
sudo systemctl disable stremio-client.service   
sudo systemctl stop stremio-server.service   
sudo systemctl stop stremio-client.service

and do 
 After the installation, try opening the Stremio server and client manually using the following   
Open first terminal and run    
```   
/home/${USER}/Stremio-RaspberryPi/src/server-launcher.sh  
```    
     
Open second terminal and run   
```   
/home/${USER}/Stremio-RaspberryPi/src/client-launcher.sh

shivasiddharth, 2020-11-24:
Its could be an issue where the client starts before the server. try stopping the service and restart the services manually. First start the server, give couple of seconds gap and start the client service.

koStelCk, 2020-11-24:
Yes it seems to be the problem.
I used
sudo systemctl stop stremio-server.service
and
sudo systemctl stop stremio-client.service

then I run
**sudo systemctl start stremio-server.service**
waited couple of seconds
run
 **sudo systemctl stop stremio-client.service**

But when I boot it again, the problem continues. Can I give a delay automatically somehow?

shivasiddharth, 2020-11-24:
you can try adding 
`sleep 10`
at this line
https://github.com/shivasiddharth/Stremio-RaspberryPi/blob/d563e97835e2c0801f9949b64255aa5ded0b7d80/src/client-launcher.sh#L8

shivasiddharth, 2020-11-24:
I have modified the service files. see if it has any effect.
Disable the services.
`sudo systemctl disable stremio-server.service`
and
`sudo systemctl disable stremio-client.service`

Then replace the existing client service file with the updated files from here https://github.com/shivasiddharth/Stremio-RaspberryPi/tree/master/systemd

Then run the system installer again and then enable the service files.
`sudo systemctl enable stremio-server.service`
and
`sudo systemctl enable stremio-client.service`

IEYTradin, 2020-11-24:
Sorry pretty new to this. How can I replace the service files?

koStelCk, 2020-11-24:
I don't know if I am doing this the right way but I just downloaded the files and replaced the new one (from downloads on the systemd folder) with the old ones ( on home/pi/Stremio-RaspberryPi). I also added on Stremio-RaspberryPi/src/client-launcher.sh the sleep 10. It works until I reboot it. maybe I have something to help. On startup I see this
[ 4.223602] systemd[1]: /lib/systemd/system/stremio-client.service:4: Failed to add dependency on network-online.target,, ignoring: Unknown error -22
[ 4.224090] systemd[1]: /lib/systemd/system/stremio-client.service:4: Failed to add dependency on stremio-server,, ignoring: Unknown error -22
[ 4.224614] systemd[1]: /lib/systemd/system/stremio-client.service:5: Failed to add dependency on network-online.target,, ignoring: Unknown error -22
[ 4.225060] systemd[1]: /lib/systemd/system/stremio-client.service:5: Failed to add dependency on stremio-server,, ignoring: Unknown error -22
[ 4.282717] systemd[1]: /lib/systemd/system/stremio-client.service:4: Failed to add dependency on network-online.target,, ignoring: Unknown error -22
[ 4.283289] systemd[1]: /lib/systemd/system/stremio-client.service:5: Failed to add dependency on network-online.target,, ignoring: Unknown error -22

shivasiddharth, 2020-11-25:
My mistake. Have fixed the files. Can you try again ?

koStelCk, 2020-11-25:
Same problem. Should I try to format the raspberry and reinstall the service?

shivasiddharth, 2020-11-25:
Reformating is not required. Just disable the previous services. Install the new updated services it should work.

koStelCk, 2020-11-28:
So it still didn't have the courtesy to work. I just made a .sh script with the following lines:

sudo systemctl stop stremio-client.service
sudo systemctl stop stremio-server.service
sleep 2
sudo systemctl start stremio-server.service
sleep 5
sudo systemctl start stremio-client.service

I looked for a tutorial on how to run a script on startup and this was the link that worked: https://www.opentechguides.com/how-to/article/raspberry-pi/5/raspberry-pi-auto-start.html
(PS if anyone will follow this link, you will see after "Auto Login" this:Read this article Raspbian Jessie/Stretch AutoLogin to Console. follow the article and then you can skip on "Run a Script after login" part)
Also after many mistakes I realized that i may wrote sleep(5) instead of sleep 5 on the line https://github.com/shivasiddharth/Stremio-RaspberryPi/issues/8#issuecomment-733007708 .. Now it works so I wont test it. :)

shivasiddharth, 2020-11-28:
I am unable to understand what you have written. In the first line you say it does not work. In the last line you say that it works.

koStelCk, 2020-11-28:
Sorry, all I mean is I installed it again and it had the same problem. So to make it work I wrote this .sh script and made it run always after login.

shivasiddharth, 2020-11-28:
Glad it got resolved.

closed 2020-11-28 by shivasiddharth