shivasiddharth/Stremio-RaspberryPi #9: Stremio stops playing

QbusKoen, 2020-11-22:
Hi there,

First of all i want to thank you for sharing this code.

But i have an issue. I can start playing a video, buffering goes fine. But after +-10s the video stops playing. Altough the video is buffered for 100%, it appears that stremio is still buffering?? When i press play, the video immadiatly starts again and then stops again after +-10.

Kind regards,

shivasiddharth, 2020-11-22:
To be 100% sure that the video has been buffered, click or move your mouse over the globe icon. That will show the amount of video that has been buffered.

QbusKoen, 2020-11-23:
Thx for your repply, but I checked the globe icon, there i see that the video is buffered for 100%.
I press play, 10s it goes fine, then i get buffering and it seems like it stays that way forever. When i press play, the video continues, again for about 10s.
Very strange...

shivasiddharth, 2020-11-23:
Which board are you using ? Also, what is the video resolution that you are trying to play ?

QbusKoen, 2020-11-23:
I first tried on a 3B+, now i'm on a 4 with 2GB. Same problem on both. For some reason i can't find the resolution settings but i'm on HDMI, i guess 1080i.
Strange thing is that video is buffered at 100% with only 512KB loaded. 
When I take a look at your server application, there i still see that it is receiving data.
I'm using TPB+ plugin by the way.

shivasiddharth, 2020-11-23:
I meant the video resolution. Whether it is 720p, 1080p or 4K ? Can you share the screenshot of the mouse over the globe icon ?

QbusKoen, 2020-11-23:
I'ts 1080p.
Sure, here they are:
[image]
[image]

shivasiddharth, 2020-11-23:
Can you try a 720p video ? Need to see if it's an issue with dropped frames.

QbusKoen, 2020-11-23:
Well, I did some extra tests. Sorry but I misunderstood you before. I thought you where referring to output resultion of the devide, not the video.
The video file had no refference wich resolution it has.
So I tried another file with 720p, which plays fine.
Then I tried another one with 1080p, which also plays fine.
The file i was playing before plays fine on my pc, There is a usergroup of 54 people, which all have the complete seasons - so i thought to continue with these files.
The only thing i can add to this is that i think it is a resolution of maybe 480p. It has the SciFi logo, so it must be recorded from a tv, early 2000.
Thank you so much for this fine piece of software! Now i gonna install HDMI-CEC to control Stremio with my IR remote... This is great!

shivasiddharth, 2020-11-24:
Can we consider this resolved ?

KoenDauwdruppel, 2020-11-25:
Yes, I just want to add that in my opinion the Rpi3B+ is not suffucient. I had much better results with model 4.
Thx again, i will enjoy using this a lot!

closed 2020-11-25 by QbusKoen