shivasiddharth/Stremio-RaspberryPi #11: Dropped frames on Pi 4?

IEYTradin, 2020-12-13:
Hi 

Many thanks for the service! 

Could It be that the frames are dropped on the Raspberry Pi 4? I have a disturbed image when playing videos. I got it on all resolutions... also when the whole file is buffered. 

Look at still image. 

Can I do something about this? 

Thanks! 
[image]
[image]

anancion, 2020-12-14:
Indeed, I got the same issue on my pi 4: very low quality image, and bad framerate.

shivasiddharth, 2020-12-14:
What resolution are you using ? Is the issue after caching or while caching ? What is the caching rate or speed ? Please mind that there are far too many factors that attribute to video playback on Pi. To test, i would suggest you to open youtube in the browser and then try and playback a 1080p video see how it goes.

IEYTradin, 2020-12-14:
I have it with both 720p and 1080p. The issue remains also after caching. 

When playing a YouTube video in 1080p I have the same problem. I tried the following solution because for many people this resolved the issues on YouTube: 

sudo rm /etc/xdg/autostart/xcompmgr.desktop 

sudo reboot

So far a little better results. 

Do you have any suggestions? 

Thanks!

shivasiddharth, 2020-12-15:
Its a Pi thing not confined to this program alone. Its going to drop frames on whatever you play in the browser, YouTube, etc. Let me check as to what can be done.

IEYTradin, 2021-01-09:
Anybody a idea on how to fix this?

shivasiddharth, 2021-01-13:
Did you check this thread ? https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=246291&start=25 It has some points there.

R3Z1H, 2021-01-17:
The videos play great on 720p (didn't test 1080p yet) but I assume it will play just fine since I can play YouTube in 1080p without any issues.

I am using Ubuntu Server 20.04 LTS and installed Ubuntu Mate desktop using **Desktopify** script by wimpysworld..

**Desktopify** does some fixes which allows hardware acceleration in Chromium.

Wimpysworld works for Ubuntu, and I think he mentioned the latest Ubuntu Desktops for RPis will support all these fixes by default in their pi OS images.

Try the latest OS by Ubuntu or use the 20.04 LTS and Desktopify it. The script I used few months ago didn't switch audio output to HDMI by default, so I had to fix that for the audio to work.

So basically if u use Ubuntu and Desktopify it, and it plays YouTube in Chromium without any issues, then this script for Stremio will work perfectly.

shivasiddharth, 2021-02-07:
I have added flags to the stremio client that should improve the playback performance considerably. Also the default compositor needs to be disabled (check README).

closed 2021-02-07 by shivasiddharth