shivasiddharth/Stremio-RaspberryPi #13: Sound not outputting

Homie-Ghost, 2021-01-03:
First of all, I just wanted to say thank you for such an amazing job with this, it truly is amazing!

Currently, my problem with this is that Stremio is not outputting any sound. It's weird b/c sound gets outputted through my TV when I go on YouTube; I'm a bit lost and confused as to why no sound outputs from the Stremio app itself.

Any and all help would be greatly appreciated :)

R3Z1H, 2021-01-17:
1) Does YouTube work for in Chromium browser or Firefox?

2) check if audio is muted for chromium browser in whatever sound manager your OS uses.

It is weird. If YouTube works in Chromium, so should Stremio..

shivasiddharth, 2021-05-04:
CLosing this for inactivity.

closed 2021-05-04 by shivasiddharth