shivasiddharth/Stremio-RaspberryPi #14: Failed to fetch

djaminou, 2021-01-21:
i have done this process. everything is ok
but when i want to watch a movies, i have a message " Failed to fetch"
Why ?

shivasiddharth, 2021-01-22:
It could be slow internet nor not enough number of seeders for a torrent stream.

dannunz1, 2021-01-31:
I am having the same issue. No streaming whatsoever. . I can search the videos but when selecting the video this message appears. I can see the same video on android with the same connection on my  PC/ android. Tried reinstalling stremio on the RPI and still getting the same issue

dannunz1, 2021-01-31:
*When selecting the torrent

dannunz1, 2021-01-31:
I have been searching and tried some fixes proposed online biut without success. I have a RPI 2 model b

dannunz1, 2021-01-31:
Cant even see  YouTube videos. Same error. And i can see YouTube via chromium without problem..

shivasiddharth, 2021-01-31:
@dannunz1 Can you check the CPU usage when you are using the stremio. Dont mind streaming, just when you have the stremio page open.

dannunz1, 2021-01-31:
I would say its pretty low, around 20%. I must say I freshly installed the new raspios
[image]

dannunz1, 2021-02-03:
Any news? I couldnt still make it work..

shivasiddharth, 2021-02-03:
I am unable to replicate the issue. What is the resolution of the YouTube videos that you say is working in Chromium ?  There will be a stats for nerds option in youtube can you share a screenshot of that at the half way mark or the video.

amrafa, 2021-02-07:
I have same issue, either in rpi4 and rpi3b+, using different network connections, cable and wifi. Also reinstalling. I would like to know if there is a way to avoid the automatic start on boot or even unistalling stremio from mi raspberry pi. Thanks and sorry for the bad news

amrafa, 2021-02-07:
during the boot there are 4 or 5 messeges saying "Failed to add dependency on..." network, stemio

amrafa, 2021-02-07:
[image]

shivasiddharth, 2021-02-07:
@amrafa 
Network wait service could be disabled and that could be giving those errors on boot.

If you wish to disable the auto start, then:
```
sudo systemctl stop stremio-client.service
sudo systemctl stop stremio-server.service

sudo systemctl dsiable stremio-client.service
sudo systemctl disable stremio-server.service
```

After disabling the service, try to manually start. To see if its an autostart issue. If you want to get rid of the stremio, just deleted the project folder after disabling the autostart.

shivasiddharth, 2021-02-07:
I have fixed this issue. Within the systemd folder inside the Stremio-RaspberryPi directory, you should find two files.
In the stremio-client.service file,

Replace:
```
Wants=network-online.target, stremio-server, graphical.target
After=network-online.target, stremio-server, graphical.target
```
With:
```
Wants=stremio-server.service graphical.target
After=stremio-server.service graphical.target
```

And then in the stremio-server.service file,

Replace:
```
Wants=network-online.target, graphical.target
After=network-online.target, graphical.target
```
With:
```
Wants=network-online.target graphical.target
After=network.target network-online.target graphical.target
```

After that,
Run:
`sudo /home/${USER}/Stremio-RaspberryPi/scripts/service-installer.sh  `

Reboot and the issue should be gone.

closed 2021-02-07 by shivasiddharth

amrafa, 2021-02-07:
I tried but the issue is still there

In the terminal I see this logg:

libEGL warning: DRI2: failed to authenticate
[1099:1099:0207/112301.555173:ERROR:viz_main_impl.cc(229)] Exiting GPU process due to errors during initialization
[1042:1172:0207/112303.182660:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1042:1172:0207/112303.184348:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1042:1172:0207/112303.185658:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1104:1109:0207/112314.054462:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -202
[1104:1109:0207/112336.471378:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.473201:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.585127:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.586421:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.867611:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.869370:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.931616:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.955549:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.966642:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.990649:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112336.992031:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.009899:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.043867:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.085547:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.112088:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.148090:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.275654:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.291947:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.293490:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.297921:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.301213:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.355814:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.391765:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.395651:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.396755:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.397782:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.398871:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.475897:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.567680:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112337.681385:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112450.458185:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -202
[1104:1109:0207/112450.458846:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -202
[... cut ...]
[1104:1109:0207/112848.180425:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112848.310423:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112848.311658:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100
[1104:1109:0207/112848.312767:ERROR:ssl_client_socket_impl.cc(960)] handshake failed; returned -1, SSL error code 1, net_error -100

Hope it is usefull and thanks for your efforts in creating this tool and resolvings its issues.

shivasiddharth, 2021-02-07:
@amrafa  What you have posted looks like is from the client terminal.  Did you have the server running when you started the client ? If you are manually launching the client, then you need to manually launch the server as well. The issue reported was only for auto start I believe, coz there is no issue with respect to manual starts.

dannunz1, 2021-02-23:
Hi, sorry for the long time without saying a thing. I think we are making progress!  Now I get this message
[image]

dannunz1, 2021-02-23:
This is on startup

commit 18b699c2 (2021-02-07): Fix auto start issue

Fixes #14