shivasiddharth/Stremio-RaspberryPi #17: Cannot find module "/stermio-shell/server.js"

connorconway, 2021-05-24:
Hello there, 

I have performed the istallation which took around 15 minutes. I have the two shortcuts for "stremio-client-launcher" and "stremio-client-service" on my desktop.

However, when i run:

```/server.js``` I get the following error:

```Error: Cannot find modile '/home/pi/Stremio-RaspberryPi/stremio-shell/server.js'```

Any ideas what is going wrong?

Thank you

connorconway, 2021-06-20:
Thank you for your response. I am still having this issue sadly :( 

Here is the terminal output:

`pi@raspberrypi:~ $ git clone http://github.com/shivasiddharth/stremio-raspberrypi

Cloning into 'stremio-raspberrypi'...
warning: redirecting to https://github.com/shivasiddharth/stremio-raspberrypi/
remote: Enumerating objects: 233, done.
remote: Counting objects: 100% (233/233), done.
remote: Compressing objects: 100% (191/191), done.
remote: Total 233 (delta 66), reused 179 (delta 34), pack-reused 0
Receiving objects: 100% (233/233), 4.94 MiB | 4.40 MiB/s, done.
Resolving deltas: 100% (66/66), done.
pi@raspberrypi:~ $ sudo chmod +x /home/pi/stremio-raspberrypi/scripts/installer.sh 
pi@raspberrypi:~ $ sudo chmod +x /home/pi/stremio-raspberrypi/scripts/service-installer.sh 
pi@raspberrypi:~ $ sudo chmod +x /home/pi/stremio-raspberrypi/src/server-launcher.sh 
pi@raspberrypi:~ $ sudo chmod +x /home/pi/stremio-raspberrypi/src/client-launcher.sh 
pi@raspberrypi:~ $ sudo /home/pi/stremio-raspberrypi/scripts/installer.sh 

Get your popcorn while we prep your Pi......

This script must run as pi, trying to change user...

Get your popcorn while we prep your Pi......

Updating the OS.......

Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [375 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Fetched 13.5 MB in 25s (542 kB/s)                                              
Reading package lists... Done

Installing required packages........

Reading package lists... Done
Building dependency tree       
Reading state information... Done
g++ is already the newest version (4:8.3.0-1+rpi2).
libmpv-dev is already the newest version (0.29.1-1).
nodejs is already the newest version (10.24.0~dfsg-1~deb10u1).
pkgconf is already the newest version (1.6.0-1).
qt5-default is already the newest version (5.11.3+dfsg1-1+rpi1+deb10u4).
qt5-qmake is already the newest version (5.11.3+dfsg1-1+rpi1+deb10u4).
qtcreator is already the newest version (4.8.2-1).
libssl-dev is already the newest version (1.1.1d-0+deb10u6+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 150 not upgraded.

Finished installing the packages........

Changing username in the service files........

Creating desktop shortcuts for starting client .............

Finished installing. Sit back and enjoy the show.............
pi@raspberrypi:~ $ /home/pi/stremio-raspberrypi/scripts/installer.sh 

Get your popcorn while we prep your Pi......

Updating the OS.......

Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done

Installing required packages........

Reading package lists... Done
Building dependency tree       
Reading state information... Done
g++ is already the newest version (4:8.3.0-1+rpi2).
libmpv-dev is already the newest version (0.29.1-1).
nodejs is already the newest version (10.24.0~dfsg-1~deb10u1).
pkgconf is already the newest version (1.6.0-1).
qt5-default is already the newest version (5.11.3+dfsg1-1+rpi1+deb10u4).
qt5-qmake is already the newest version (5.11.3+dfsg1-1+rpi1+deb10u4).
qtcreator is already the newest version (4.8.2-1).
libssl-dev is already the newest version (1.1.1d-0+deb10u6+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 150 not upgraded.

Finished installing the packages........

Changing username in the service files........

Creating desktop shortcuts for starting client .............

Finished installing. Sit back and enjoy the show.............
pi@raspberrypi:~ $ /home/pi/stremio-raspberrypi/src/server-launcher.sh 
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module '/home/pi/Stremio-RaspberryPi/stremio-shell/server.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:623:3)
`

Any ideas on what I should try next?
Thank you

fex0r, 2021-07-05:
Installing the latest node version should properly fix this error
If you don't know how to install node 16 on raspy, just [follow this guide](https://thisdavej.com/beginners-guide-to-installing-node-js-on-a-raspberry-pi/)

connorconway, 2021-07-20:
The error is subtly different now, but still the same:

```
pi@raspberrypi:~ $ sudo /home/pi/stremio-raspberrypi/src/server-launcher.sh 
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module '/home/pi/Stremio-RaspberryPi/stremio-shell/server.js'
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader.js:927:15)
    at Function.Module._load (node:internal/modules/cjs/loader.js:772:27)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:79:12)
   at node:internal/main/run_main_module:17:47 {
code: 'MODULE_NOT_FOUND',
requireStack: []
}
```

I can launch client-launder.sh successfully, but when clicking any movie to watch it fails to fetch. 

Really appreciate any further assistance as I'm scratching my head here on this one. Thank you both @fex0r  @shivasiddharth

shivasiddharth, 2021-07-20:
I am right now busy with personal agenda. Will get back on this in 2 days.

connorconway, 2021-07-20:
Appreciate the help and thank your time on this :)

shivasiddharth, 2021-07-24:
You need to be on Node version 10. Just tested it and it works.

connorconway, 2021-08-06:
I really appreciate your help on this @shivasiddharth, but it still does not work for me. :( 

If it helps I a running Raspberry Pi 4 4GB with Ubuntu 

`pi@raspberrypi:~ $ sudo /home/pi/stremio-raspberrypi/src/server-launcher.sh 
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module '/home/pi/Stremio-RaspberryPi/stremio-shell/server.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Function.Module.runMain (internal/modules/cjs/loader:829:12)
   at startup (internal/bootstrap/node.js:283:19)
   at bootstrapNodJSCore (internal/bootstrap/node.js:622:3)

pi@raspberrypi:~ $ node -v
10.16.0
`

At this point I would be happy to pay for any specific help you can provide as I don't see what is going wrong.

Thanks @shivasiddharth

ncs92, 2021-08-13:
@connorconway The problem is probably because your project is not in the "home/pi" directory

shivasiddharth, 2021-08-13:
Ok just noticed something. The folder naming is slightly different in your case. 

You have cloned the git as ```stremio-raspberrypi``` while the command in the bash files have it as ```Stremio-RaspberryPi```. 

The script is looking for the server file in ``` '/home/pi/Stremio-RaspberryPi/stremio-shell/``` while you have it in  ```'/home/pi/stremio-raspberrypi/stremio-shell/```.  Something so basic and yet easy to miss and overlook.

shivasiddharth, 2021-08-25:
Closing this for inactivity.

closed 2021-08-25 by shivasiddharth