shivasiddharth/Stremio-RaspberryPi #18: Failed to fetch

JexxeL, 2021-05-27:
Hello ShivaSiddarth.

Thank you so much for making this. I just installed it, and the client pops up! Wonderful!
However, whenever I try to stream anything, the client says "failed to fetch"  and just doesn't play.
Doesn't matter which stream I try. I started the server (through the desktop shortcut) and after that I start the client.
I already rebooted my computer, but it doesn't make any difference.

What should I do?

Thanks in advance,

JexxeL

shivasiddharth, 2021-05-28:
I fixed this issue long back. This will arise only if the client starts before the server or if the server is not running. 
Which board are you using ? 
Can you launch the server and client manually. The server should be launched first. The commands are in the README. Give about 30 seconds gap between launching the server and client. See if it works. If the manual start does jot work, please post the contents of the respective terminals here for me to look at.

JexxeL, 2021-05-28:
Thank you very much for your response! I use a Raspberry Pi 4b 4GB.
I don't think I waited 30 second between launching the server and the client.
I will try again tonight, and report back, thanks again.

jonathandalves, 2021-06-15:
Change the startup dependencies in `/lib/systemd/system/stremio-server.service` to `multi-user.target` in the `Wants` and `After` parameters.

It will look like this:

```

[Unit]
Description=Stremio Media Server
PartOf=graphical-session.target
Wants=multi-user.target
After=multi-user.target

[Service]
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/Stremio-RaspberryPi/src/server-launcher.sh

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
User=pi
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

shivasiddharth, 2021-11-13:
Not relevant with the new update.

closed 2021-11-13 by shivasiddharth