shivasiddharth/Stremio-RaspberryPi #20: 123movies can't play 1080p, but 720p works fine

Dyras, 2021-06-29:
Hello! I've noticed that for whatever reason 123movies can't play 1080p content, for seemingly any movie.

720p on the other hand works almost instantly, ever time.

Anyone else with a similar problem?

shivasiddharth, 2021-07-07:
Check the number of SEEDS. I have no control of what plays what. Using the ```sudo raspi-config``` increase the GPU memory if you have Pi4.  Install H264ify plugin in chromium. Last suggestion is to use VLC.

shivasiddharth, 2021-07-09:
You can remote access Stremio server on Apple devices/Windows/Android. Open a browser and type in https://IP-ADDRESS-OF-PI:12470 This will openup stremio on other devices. You need not install apps individually in all your devices. For ref: https://www.youtube.com/watch?v=cYmpVh_GWJg

Dyras, 2021-07-10:
Haven't noticed any issues with the seeds. The exact same videos work on Windows.

Will try to increase the GPU memory though. Feeling dead this weekend so I'll test about on Monday or so.

shivasiddharth, 2021-11-13:
Fixed !

closed 2021-11-13 by shivasiddharth