shivasiddharth/Stremio-RaspberryPi #21: Latest Server.js does not support Remote Access

ashwarsadh, 2021-07-02:
In Older Server Version 4.2.3, we can just forward port 11470 , then open http://app.strem.io/shell-v4.4/#/ on any browser on any device like non android Smart TV, iOs etc, and enter server url in settings and Stremio used to work fine.
In newer Version, it seems remote access is only enabled for Https endpint, having port 12470, but Node https Server fetches certificate for localhost only and if we try accessing from remote IP, it does let access due to certificate mismatch.
In Desktop Clients, https url is shown, but it also only works in LAN network, unless server is directly exposed to internet , having a public IP.
So, please provide a workaround so that a url with proper ssl certificate is generated which Stremio browser url can access remotely as well.
As of now, workaround is to use older version f server, but Stremio web client may depreciate it eventually.

ashwarsadh, 2021-07-02:
sudo bash Stremio-RaspberryPi/src/server-launcher.sh

hls executables located ->  { ffmpeg: '/usr/bin/ffmpeg', ffsplit: null }
Using app path -> /root/.stremio-server
Enabling casting...
Discovery of new external device "mpv" - MPV
(node:4173) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(Use `node --trace-deprecation ...` to show where the warning was created)

EngineFS server started at http://127.0.0.1:11470
EngineFS HTTPS endpoint at **https://local.strem.io:12470**
HTTPS: Request error Could not get a valid HTTPS certificate
HTTPS: Request error Could not get a valid HTTPS certificate
HTTPS: Request error Could not get a valid HTTPS certificate
HTTPS: Request error Could not get a valid HTTPS certificate

Above errors appear when trying to remotely connect from https://url:12470 and http server does not work only

shivasiddharth, 2021-07-03:
Can you try running it without sudo ? The commands that I have given here does not include the sudo https://github.com/shivasiddharth/Stremio-RaspberryPi#after-the-installation-try-opening-the-stremio-server-and-client-manually-using-the-following

shivasiddharth, 2021-07-03:
Are you using an apple device ? My android devices are all working fine.

ashwarsadh, 2021-07-03:
I have tried on Google Chrome on Windows pc, Android, tv browsers etc.
After entering server url when I plan any stream it is unable to play.
On rolling back to older server.js file, everything works fine.

shivasiddharth, 2021-07-03:
I have created SSL certificates and now issue has been fixed.

shivasiddharth, 2021-07-03:
Delete the old project folder and clone again. I have tested it on all platforms and it seems to work.

commit ad49c717 (2021-07-03): Fix SSL Remote Access

Fixes #21
closed 2021-07-03 by shivasiddharth