shivasiddharth/Stremio-RaspberryPi #22: Dynamic AppPath causes unable to find sslcerts folder

ashwarsadh, 2021-07-03:
Hello,
 Latest update did correct SSL issue, but I had to manually copy sslcerts folder to /home/$user/.stremio-server/ 
AppPath (Working Directory) changes in server.js , it first looks for /.stremio-server/ folder, if not found then switches to stremio-shell directory.
Line 133749 of server.js
appPath = path.join(process.env.HOME, "."+pkg.name);

so a new user may face errors as below:
/Stremio-RaspberryPi/src/server-launcher.sh

hls executables located ->  { ffmpeg: '/usr/bin/ffmpeg', ffsplit: null }
Using app path -> /root/.stremio-server
Enabling casting...
Discovery of new external device "mpv" - MPV
node:internal/fs/utils:343
    throw err;
    ^

Error: ENOENT: no such file or directory, open './sslcerts/ca.key'
    at Object.openSync (node:fs:582:3)
    at Object.readFileSync (node:fs:450:35)
    at Object.require.720../cache (/root/Stremio-RaspberryPi/stremio-shell/serve                               r.js:134024:44)
    at s (/root/Stremio-RaspberryPi/stremio-shell/server.js:1:338)
    at e (/root/Stremio-RaspberryPi/stremio-shell/server.js:1:509)
    at Object.<anonymous> (/root/Stremio-RaspberryPi/stremio-shell/server.js:1:5                               27)
    at Module._compile (node:internal/modules/cjs/loader:1095:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1124:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:816:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_ma                               in:79:12)
    at node:internal/main/run_main_module:17:47 {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT',
  path: './sslcerts/ca.key'
}

shivasiddharth, 2021-07-03:
Its searching for that in the wrong folder, because you running it as a root user with sudo. I have clearly instructed to not run the commands as root user.  
`/root/Stremio-RaspberryPi/stremio-shell/serve r.js:134024:44)`
From the looks of it, looks like you have used sudo for even cloning the git.  Why ????

closed 2021-07-03 by shivasiddharth

ashwarsadh, 2021-07-03:
Ok, will install in pi User.
Stremio uses by default 2.5MBPS as download speed cap, we have much faster
internet connection and want it to utilise more speed, so Stremio advises
following way:
https://blog.stremio.com/modify-bittorrent-settings/

"

   1. In the terminal window type „mkdir ~/.stremio-server” and press
   enter. Now .stremio-server folder is created.
   2. Now type “touch ~/.stremio-server/server-settings.json” and press
   enter. The server-settings.json file is created. "

   After doing above steps, Working directory changes to  ~/.stremio-server
   and stremio will stop working as it wont be able to find sslcerts folder

   But I do agree, running as root/sudo is my mistake, and if strictly
   following your instructions everything shall work fine