shivasiddharth/Stremio-RaspberryPi #23: Using h264_omx Encoder for Faster Transcoding on rPi

ashwarsadh, 2021-07-03:
Stremio uses libx264 encoder in ffmpeg to transcode .mkv streams, this codec is univerally compatible but uses CPU power, which is also being used for torrent downloading simultaneosly, leading to 100% cpu usage, video stuttering and even cpu throttling on our Raspberry pi.

h264_omx is better for Raspberry Pi as it uses GPU Instead, so it is much faster, Stremio does not use it by default due to inter-device compatibility reason.
If we add a check that if device is an rPi then use h264_omx then this may improve mkv streaming performance.

shivasiddharth, 2021-07-03:
Will try and see as to what can be done.

shivasiddharth, 2021-07-06:
h264_omx has negligible effect. Just using h264_omx does not enable hardware acceleration. You need to enable hardware acceleration in stremio settings as well. But even that has hardly any effect. If you are watching directly on Pi, right click on the video frame and choose to play using VLC. That will reduce the CPU usage. If you are using Pi only as a server, the CPU usage not very high even if you stream from multiple devices.

ashwarsadh, 2021-07-07:
Have you tried when playing any mkv file, which requires Transcoding.
While playing such file it shows Warning:Transcoding under stats and uses
100% CPU, specially if 1080p content causing stuttering in playback

shivasiddharth, 2021-07-07:
I have not encountered any issue that you are saying. Still I have changed the codec to h264_omx. I would strictly suggest you to use VLC. That offers excellent clarity and reduced CPU usage. First install VLC and then when the video is playing, right click on the video and choose to play in VLC. I have added even support for subtitles.

closed 2021-07-07 by shivasiddharth