shivasiddharth/Stremio-RaspberryPi #25: Can't play videos in VLC

Dyras, 2021-07-07:
Hello! Sorry for being an annoyance, but for whatever reason I can't get videos to play in VLC.

I click a video in Torrentio and wait for the video to start. Once it's playing, I click the icon in the bottom right corner and choose "Play in VLC"

Then Stremio goes back to loading, and VLC doesn't open. VLC came installed in my Raspberry Pi OS install, but I tried reinstalling it and no success so far.

I've also tried non-Torrentio files, like one from 123movies.

Raspbian GNU/Linux 10 (buster)
Latest Stremio, updated 10 minutes ago

Any help appreciated.

shivasiddharth, 2021-07-07:
Are you interacting with your Pi using the VNC remote desktop service? Try connecting a monitor.

Dyras, 2021-07-07:
When I tried using a keyboard and mouse instead of VNC (the Pi is connected to the tv) it instantly worked

Now I'm almost afraid to ask... how do I pause the video in VLC? I can't see it being open at all in VNC and pressing on the keyboard does nothing :D 

Thanks for the help!

shivasiddharth, 2021-07-07:
I am here just to help ! VLC and all other media players have some weird issue. If used with Stermio service that has been setup to auto start on boot, it is loosing all controls. The way to get around is to stop the services and manually launch. To stop the services use:
```sudo systemctl disable stremio-server.service```
```sudo systemctl disable stremio-client.service```
To manually launch, use the commands given [here](https://github.com/shivasiddharth/Stremio-RaspberryPi#after-the-installation-try-opening-the-stremio-server-and-client-manually-using-the-following) 

I have updated the README to reflect these. 

Delete the old shorcuts on the desktop. Open a terminal and run the following. These will create two new shorcuts. Using this you can manually launch the service and client.

```
sudo chmod +x /home/${USER}/Stremio-RaspberryPi/src/server-launcher.sh    
sudo chmod +x /home/${USER}/Stremio-RaspberryPi/src/client-launcher.sh  
sudo \cp /home/${USER}/Stremio-RaspberryPi/src/server-launcher.sh /home/${USER}/Desktop/Stremio-Server-Launcher-Shortcut
sudo \cp /home/${USER}/Stremio-RaspberryPi/src/client-launcher.sh /home/${USER}/Desktop/Stremio-Client-Launcher-Shortcut
```

Dyras, 2021-07-07:
Yeah I saw your edit of the "manual", thanks for that! I thought I was going insane.
Hmm, is there a way to make it auto start without this weird issue? I tried CronTab but it seems to have the same issue.

shivasiddharth, 2021-07-07:
I have been breaking my head over this ever since the project was launched. I am close to fixing it.

shivasiddharth, 2021-07-07:
I have rolled out the fix for VLC issue. To move from old to the updated version:
1. Disable old services using (they are no longer relevant): 
```
sudo systemctl disable stremio-server.service
sudo systemctl disable stremio-client.service
```
2. Delete the old project folder and existing shortcuts on the desktop.
3. Re-clone the git, and follow the setup commands given in the README. 
4. This is a major bump-up so you need to start from [here](https://github.com/shivasiddharth/Stremio-RaspberryPi#make-the-scripts-executable) and proceed as though starting fresh.

closed 2021-07-07 by shivasiddharth

Dyras, 2021-07-07:
@shivasiddharth if I wanted to sponsor you with a few bucks every month, how would I go about doing so?

shivasiddharth, 2021-07-08:
@Dyras If you wanted to sponsor, then head over to the project's home page https://github.com/shivasiddharth/Stremio-RaspberryPi at the top you should see a sponsor button. Click on that and follow the instructions.

rimthekid, 2021-11-16:
@shivasiddharth Could you update the code too I just build from it and opening from vlc didn't work also why is there a very low frame rate playing inside the app but it's ok when playing with vlc

shivasiddharth, 2021-11-16:
@rimthekid  Which board and OS are you on ?  The new build works brilliantly even within the interface. Check the demo here https://youtu.be/ZGXpppWgttE The VLC is also enabled by default.

PR #26 (2021-07-07): VLC Issues Fixes
Fixes #25  Fixes #1 
1. Services are run as user and not as root user.
2. VLC and other media player controls work even for services.
3. A minor delay has been introduced to the start of client service.