shivasiddharth/Stremio-RaspberryPi #28: Stremio server service does not start on Raspbian Stretch

sudogitclone, 2021-07-23:
Hello,

Stremio server service does not start on my Raspberry Pi 3B+ running Raspbian Stretch:

```
~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs
```

It seems the [Stremio server service unit](https://github.com/shivasiddharth/Stremio-RaspberryPi/blob/master/systemd/stremio-server.service) uses a different path to bash:

```
ExecStart=/usr/bin/bash /home/__USER__/Stremio-RaspberryPi/src/server-launcher.sh
```

Where it should be:

```
~ $ which bash
/bin/bash
```

Editing the bash path in the unit file to `/bin/bash` allows the server service to start normally. This is also the same in the client service.

sudogitclone, 2021-07-24:
On further observation I noticed after enabling the Stremio server service, the user@1000 service started failing to start after a reboot which breaks all user systemd services from starting (similar to [this issue](https://bbs.archlinux.org/viewtopic.php?id=227446) and [this](https://bugs.archlinux.org/task/54396)). Prior to installing the service my /etc/systemd/user was empty. I fixed it by deleting the 2 Stremio service files and rebooting. I now autostart the server using @reboot cron job, but I suppose it would be better to put the service files in ~/.config/systemd/user/.

shivasiddharth, 2021-07-24:
All the mods in this project are for Buster and beyond. Buster by itself is close to 2 years old and we are almost close to the launch of Bullseye. I don't think I can support mods for older versions. Users on older versions have to figure out issues on their own. Its a one man's work and its impossible for me to test the installation in all the Raspi OS versions. Test your mod on buster and if it works create a pull request, I am happy to merge.

sudogitclone, 2021-07-25:
@shivasiddharth that is understandable, so far everything works well in Stretch so I will close this issue. Thanks for the good work!

closed 2021-07-25 by sudogitclone