shivasiddharth/Stremio-RaspberryPi #29: Always transcoding, even if file resolution is lower than screen.

iriki, 2021-10-06:
Hi, I have an issue. When streaming video, it's very laggy and it shows a lot of compression artifacts. Is it possible (and where) to completely disable the transcoding? 

I have a Rpi400 4Gb. I can see a lot of activity of ffmpeg if I run htop, and IMHO there shouldn't be any reason why transcoding happens  because I'm connected to a 1920x1080 TV and the source file has the same resolution. I made a small test, I've downloaded the very same file and played it locally in VLC, absolutely no issues, very fluid, no frame drops and no stutters. 

I've tried streaming a show with 480p and I can still see ffmpeg activity in the console. 

So I just want to disable transcoding for now, in order to better troubleshoot the issue. Many thanks in advance

iriki, 2021-10-06:
You may be on to something: in VLC it plays in full quality and it doesn't skip frames, however there are no subtitles available :(

shivasiddharth, 2021-10-08:
Its an issue with the browser itself. Pi can't render even YouTube 1080p videos clean. Still if you want to check the server file you can use ATOM.

shivasiddharth, 2021-11-13:
Fixed !

closed 2021-11-13 by shivasiddharth