shivasiddharth/Stremio-RaspberryPi #30: Unable to Install

lmjlopes, 2021-11-11:
Hi,

Many thanks for the great work!

I'm experiencing an error after issuing "sudo /home/${USER}/Stremio-RaspberryPi/scripts/installer.sh"

Package qt5-default is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'qt5-default' has no installation candidate

Can you help on this issue?

Thanks.

shivasiddharth, 2021-11-11:
Which board and OS version are you using ?

lmjlopes, 2021-11-11:
Raspbian GNU/Linux 11 (bullseye).

Trying to setup only the server part on a recently bought pi zero 2w, to be accessed remotely from an ipad, as described in https://www.youtube.com/watch?v=cYmpVh_GWJg

shivasiddharth, 2021-11-11:
Bullseye does not have the qt5 package. You need to be on Stretch.

lmjlopes, 2021-11-11:
thanks @shivasiddharth for the fast response and clarification.

lmjlopes, 2021-11-11:
Just flashed with Stretch but Pi Zero 2w won't boot with it. Just when reflashed with Bullseye the Pi was alive again.
Any ideas to overcome this?

shivasiddharth, 2021-11-11:
I don't have a Pi Zero 2 W with me. Can you check in the Raspberry Pi forums if the Stretch is compatible or not ? In any case, I will be releasing compiled binaries in the coming days. That should work in any build.

shivasiddharth, 2021-11-12:
Ok, gave your issue some thought. Try this.
Load up the working Bullseye on your Pi 2 W. 

1. After it boots, do:
`sudo nano /etc/apt/sources.list`
In that, add the following
```
deb http://ftp.us.debian.org/debian/ buster main contrib non-free
deb-src http://ftp.us.debian.org/debian/ buster main contrib non-free
```
2. After that, do:
`sudo apt-get update`

3. Now try this project.

4. After the project starts working, open the sources list again and remove the parts that you earlier added.

lmjlopes, 2021-11-12:
Hi @shivasiddharth, many thanks for your interest in this case.

I'll explain what I did meanwhile. It was a little "brute force", but I managed to install the server and run it.

I installed all dependencies of qt5-defalt package with:
sudo apt-get install qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools

then, I "faked" the missing package with:
cat <<EOF > qt5-default-control
Package: qt5-default
Source: qtbase-opensource-src
Version: 5.99.99+fake-13ubuntu37
Architecture: all
Depends: qtbase5-dev, qtchooser
Suggests: qt5-qmake, qtbase5-dev-tools
Conflicts: qt4-default
Section: libdevel
Priority: optional
Homepage: http://qt-project.org/
Description: Qt 5 development defaults fake package
EOF

then:
sudo apt-get install equivs
equivs-build qt5-default-control
sudo apt-get install ./qt5-default_5.99.99+fake-13ubuntu37_all.deb

I was then able to install successfully and connect with a bowser to the server.

Nevertheless, I'm experiencing some intermittent results. When I'm able to open a source, it works really well, 1080p resolution, good audio, no lagging, CPU on the rpizero 2w around 20%-40% (provided by another terminal with top). I was able to play 5 or 6 different sources now.

But if I try to open another different source, it seems to cache it, but then never starts playing, giving an error of low sources or low internet speed. This happens to lots of sources.

Do you think this behavior is related with the installation process I used? If yes, I can start from scratch and try your suggestion.

Many thanks.

lmjlopes, 2021-11-12:
an addon:
The sources that won't play work really well on an android based stremio, so there are no issues with sources or internet speed.

lmjlopes, 2021-11-12:
when it doesn't start, I see this on the terminal:

Unknown schemaID  0x55b0 tagId= 41 start=589 length=2 source= HttpSource {
  _vints: [],
  _vintsSize: 0,
  configuration: {},
  url: 'http://127.0.0.1:11470/d4c0ebc41779265048087d54306d614636953438/0'
}

shivasiddharth, 2021-11-12:
Faking could be an issue, but as I dont have a first hand experience, I cannot speculate on it. I would suggest you to try this https://github.com/shivasiddharth/Stremio-RaspberryPi/issues/30#issuecomment-966771737 If you can undo the faking, you can use the current build. Else, I would suggest you to re-image the sd card.

If this project is not a deal breaker, I would suggest you to hold back for just another two days. I have created binaries that you could use to install Stremio like any other app. The performance has also improved. I am right now testing stuff out and preparing the documentation for the it.

lmjlopes, 2021-11-12:
I re-image the sd card and followed the suggested process.

Just FYI, it throw the following error:
The following packages have unmet dependencies:
 qtbase5-dev : Breaks: qt5-default (< 5.15.1+dfsg-2~) but 5.11.3+dfsg1-1+deb10u4 is to be installed

Will just wait for your next release, as suggested.

Many thanks again for your efforts and for sharing your work.

closed 2021-11-12 by lmjlopes

shivasiddharth, 2021-11-13:
Please check the new master. The performance has improved manifold times. Follow the instructions and use the new binaries.