shivasiddharth/Stremio-RaspberryPi #32: Headless, server only - Is it possible with this release?

lmjlopes, 2021-11-13:
Hi @shivasiddharth,

The installation process went smoothly, thanks for detailed notes.

I'm trying to use stremio in an headless RPI Zero 2W, in a server only mode to be accessible via browser from Ipad.

Is this possible with this release?

When I try to start the app via /bin/stremio, it throws the following error:
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.

Thanks.

shivasiddharth, 2021-11-14:
Yes it's possible. Just type stremio in the terminal.  Preceeding /bin is not required.

shivasiddharth, 2021-11-14:
Just a point to note. For the secure https endpoint, you need to start the interface from the pi itself. Then turn on remote access and get the secure endpoint url like the one shown in the Readme Notes. The files are located in the /opt directory. Use node to start just the server for headless operation.

lmjlopes, 2021-11-14:
I was trying to keep it as light as possible, so I just flashed "Raspberry Pi OS Lite" and tried to do everything headless from SSH.
I was able to start the server via node but it obviously gave me a HTTPS certificate error when tried to connect to 12470.
What I understand is that I need to flash with "Raspberry Pi OS with desktop" and enable the remote access https endpoint with a display connected. There is no way to do it from the SSH interface, right?

Thanks again for your help.

shivasiddharth, 2021-11-14:
Exactly.

lmjlopes, 2021-11-14:
All ok, until the point where, after starting server with node (headless) it gives always the same error when trying to open a stream:

-> GET /178e8a1e82a5af2c590213c28ad26b8a1553f25a/1/hls.m3u8
/opt/stremio/server.js:134964
        var body = JSON.stringify({ error: error, result: result });
                        ^

TypeError: Converting circular structure to JSON
    --> starting at object with constructor 'Object'
    --- property 'issuerCertificate' closes the circle
    at JSON.stringify (<anonymous>)
    at /opt/stremio/server.js:134964:18
    at /opt/stremio/server.js:140589:21
    at /opt/stremio/server.js:135257:25
    at ClientRequest.<anonymous> (/opt/stremio/server.js:125490:21)
    at ClientRequest.f (/opt/stremio/server.js:106176:25)
    at ClientRequest.emit (events.js:314:20)
    at ClientRequest.EventEmitter.emit (domain.js:483:12)
    at TLSSocket.socketErrorListener (_http_client.js:427:9)
    at TLSSocket.emit (events.js:314:20)
    at TLSSocket.EventEmitter.emit (domain.js:483:12)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21)

is it something with my setup?

Thanks.

shivasiddharth, 2021-11-14:
Looks like an issue with the secured endpoint. Does it affect the stream on the remote device. Stremio is blocking the changes that I make to the server file. So cant do anything more. The last suggestion for you would be to try streaming on the remote device with the interface running on the Pi.

shivasiddharth, 2021-11-22:
Do you have an update on this issue ?

lmjlopes, 2021-11-22:
Hi @shivasiddharth, 

It does affect the stream, I cannot play any content.
Since I wasn't able to get this working as intended (RPI doing streaming, Ipad as the viewing interface), I'm now seeking for a different solution to achieve my goal.

Thanks again for your help trying to solve this.

closed 2021-11-22 by lmjlopes