shivasiddharth/Stremio-RaspberryPi #33: Unable to Install Raspbian - Debian Buster 10.11 based

iriki, 2021-11-23:
Hello. I'm on Raspbian Debian Buster:

> pi@raspberrypi:~/stremio-shell $ cat /etc/debian_version 
10.11

...  but I'm having issues installing the newer version:

> pi@raspberrypi:~/stremio-shell $ sudo dpkg -i /tmp/Stremio-4.4.142-armhf-32-bit/stremio_4.4.142-1_armhf.deb 
Selecting previously unselected package stremio.
(Reading database ... 115112 files and directories currently installed.)
Preparing to unpack .../stremio_4.4.142-1_armhf.deb ...
Unpacking stremio (4.4.142-1) ...
dpkg: dependency problems prevent configuration of stremio:
 stremio depends on qml-module-qt-labs-platform (>= 5.9.5); however:
  Package qml-module-qt-labs-platform is not installed.
 stremio depends on qml-module-qtquick-dialogs (>= 5.9.5); however:
  Package qml-module-qtquick-dialogs is not installed.
 stremio depends on qml-module-qtwebchannel (>= 5.9.5); however:
  Package qml-module-qtwebchannel is not installed.
 stremio depends on qml-module-qtwebengine (>= 5.9.5); however:
  Package qml-module-qtwebengine is not installed.
 stremio depends on qml-module-qt-labs-folderlistmodel (>= 5.9.5); however:
  Package qml-module-qt-labs-folderlistmodel is not installed.
 stremio depends on qml-module-qt-labs-settings (>= 5.9.5); however:
  Package qml-module-qt-labs-settings is not installed.
 stremio depends on libqt5webview5-dev; however:
  Package libqt5webview5-dev is not installed.
 stremio depends on libkf5webengineviewer-dev; however:
  Package libkf5webengineviewer-dev is not installed.
>    
> dpkg: error processing package stremio (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 stremio
pi@raspberrypi:~/stremio-shell $ 

> pi@raspberrypi: ~ /stremio-shell $ sudo apt autoremove
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 stremio : Depends: qml-module-qt-labs-platform (>= 5.9.5) but it is not installed
           Depends: qml-module-qtquick-dialogs (>= 5.9.5) but it is not installed
           Depends: qml-module-qtwebchannel (>= 5.9.5) but it is not installed
           Depends: qml-module-qtwebengine (>= 5.9.5) but it is not installed
           Depends: qml-module-qt-labs-folderlistmodel (>= 5.9.5) but it is not installed
           Depends: qml-module-qt-labs-settings (>= 5.9.5) but it is not installed
           Depends: libqt5webview5-dev but it is not installed
           Depends: libkf5webengineviewer-dev but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
pi@raspberrypi:~/stremio-shell $ 

Note that I added 

`
deb http://ftp.us.debian.org/debian/ buster main contrib non-free
deb-src http://ftp.us.debian.org/debian/ buster main contrib non-free
`

as stated on the ticket #30 .

Can anyone help? Many thanks in advance.

shivasiddharth, 2021-11-23:
A couple of issues. Looks like you have not followed the README completely. 
1. Why do you have a stremio-shell folder ?
2. The installation command is not what I have given. (Did you see 5. under https://github.com/shivasiddharth/Stremio-RaspberryPi#installation-using-binaries ?)
3. Why is the binary in the tmp folder ?

Getting stremio to work on Pi is tricky. It took me over 50 attempts to figure out the exact step-by-step procedure. If you deviate from the exact steps, its not going to work.

closed 2021-11-23 by shivasiddharth

iriki, 2021-11-29:
Yay it's working, I feel ashamed and I'm sorry I haven't followed all the instructions thoroughly, now it's working, albeit it presents a bit tearing when there's a lot of horizontal movement.

shivasiddharth, 2021-11-29:
### Fix for screen tearing    
Open a terminal and run   
```    
sudo raspi-config     
```   
In that, Navigate to Advanced Options -> Compositor -> xcompmgr composition manager -> Choose “No”     

Make sure to increase the GPU memory also.

shivasiddharth, 2021-11-29:
You need to do a restart after that.

iriki, 2021-12-09:
Thanks, but xcompmgr composition manager was already disabled on my Buster installation.

After much fiddling, I found the cause (and also the fix!):

Chromium had Video acceleration disabled. Go to `chromium://gpu` and check under "Video decode". Mine was on software rendering. So I found [this tutorial](https://lemariva.com/blog/rss/raspberry-pi-4-video-acceleration-decode-chromium) and followed the steps, but unfortunately on Fig.5 while trying to activate `Hardware-accelerated video decode`, that option was  "Not available on your platform".

After more searching, I found out that starting chromium with the flag `--enable-features=VaapiVideoDecoder` the hardware  video decoding was now enabled in  `chromium://gpu`. Woh-woh! 

Now I needed a way to pass that flag back to Stremio. After even more searching, I found that one can pass that command line options inside ~/.chromium-browser.init. This way it will be picked up by all Chromium instances.

CHROMIUM_FLAGS="--enable-features=VaapiVideoDecoder"

Now when starting Chromium normally, the `Video Decode: Hardware accelerated` should now appear in Green and ready to rock! The video playback on Stremio is now butter-smooth and I can no longer see any tearing whatsoever. Hope this helps anyone with the same issue.

iriki, 2021-12-22:
Hello again. After reinstalling Debian Buster 2021-12-02 version, the upper fix no longer worked. After much fiddling, the flags now need to be set in another location, this time at the end of the file `/etc/chromium.d/default-flags`, like so:

```
export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --enable-features=VaapiVideoDecoder"
export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --ignore-gpu-blocklist"
```
Now the flags will be picked up automatically by any instance of chromium, including the one ran by Stremio. Enjoy flicker-free experience!