shivasiddharth/Stremio-RaspberryPi #35: Issue when doing sudo apt-get install ./libfdk-aac1_0.1.6-1_armhf.deb ./stremio_4.4.142-1_armhf.deb

jpants36, 2022-03-04:
When I do this, I get an error that says:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 stremio : Depends: qml-module-qtwebengine (>= 5.9.5) but it is not installable
           Depends: libqt5webview5-dev but it is not installable
           Depends: libkf5webengineviewer-dev but it is not installable
E: Unable to correct problems, you have held broken packages.

shivasiddharth, 2022-03-04:
Share the entire screen contents. I would like to see the command that you issued as well. In addition, provide me with the output of:
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```

jpants36, 2022-03-04:
cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 00000000ee3a6e2c
Model		: Raspberry Pi 3 Model B Rev 1.2

cat /etc/rpi-issue
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 60f6a26ed5701eec6be5c864dd0db48fe6244fad, stage4

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

shivasiddharth, 2022-03-04:
You have answered only a part of my query. I want to see the entire terminal contents when you start the installation.

jpants36, 2022-03-04:
Sorry. Just re-done it and found it was the pubkey error thing. I will let you know if I run into any more issues

shivasiddharth, 2022-03-05:
I have taken utmost steps to make it easy for everyone. The PUBKEY warning ahs also been addressed in the notes. Pi3 will not offer the same performance as Pi4 for media streaming. Just a word of caution.

jpants36, 2022-03-16:
I am once again experiencing issues. Here is my terminal when I try to install it:
pi@raspberrypi:~ $ sudo sh -c "echo 'deb http://ftp.us.debian.org/debian/ buster main contrib non-free' >> /etc/apt/sources.list"   
    sudo sh -c "echo 'deb http://deb.debian.org/debian buster main contrib non-free' >> /etc/apt/sources.list"     
    sudo apt-get update
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                
Hit:3 http://deb.debian.org/debian bullseye InRelease                          
Hit:4 http://archive.raspberrypi.org/debian buster InRelease                   
Hit:5 http://ftp.us.debian.org/debian buster InRelease                         
Ign:6 https://repo.vivaldi.com/stable/deb stable InRelease                     
Hit:7 https://repo.vivaldi.com/stable/deb stable Release                       
Hit:8 http://ftp.us.debian.org/debian bullseye InRelease                       
Ign:9 https://download.puffinbrowser.com/repo stable InRelease                 
Hit:10 https://download.puffinbrowser.com/repo stable Release
Reading package lists... Done
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Packages (contrib/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Packages (contrib/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Translations (contrib/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Translations (contrib/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Packages (non-free/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Packages (non-free/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Translations (non-free/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Translations (non-free/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:9
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Packages (contrib/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Packages (contrib/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Translations (contrib/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Translations (contrib/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Packages (non-free/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Packages (non-free/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Translations (non-free/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Translations (non-free/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:11
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
W: Target Packages (contrib/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
W: Target Packages (contrib/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
W: Target Translations (contrib/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
W: Target Translations (contrib/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:13
[... cut ...]
Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 stremio : Depends: libmpv1 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

shivasiddharth, 2022-03-16:
give me the output of:
`sudo nano /etc/apt/sources.list`

shivasiddharth, 2022-03-16:
you need only  1 instance of 
```
deb http://ftp.us.debian.org/debian/ buster main contrib non-free
deb http://deb.debian.org/debian buster main contrib non-free
```
The first command should be run only once. remove the rest and save the file and give the sudo apt-get update

jpants36, 2022-03-16:
Done. Just tried to install again now to see if it would actually install, but now I'm getting a different issue.

pi@raspberrypi:~/Stremio-4.4.142-armhf-32-bit $ sudo apt-get install ./libfdk-aac1_0.1.6-1_armhf.deb ./stremio_4.4.142-1_armhf.deb -f
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
libfdk-aac1 is already the newest version (0.1.6-1).
stremio is already the newest version (4.4.142-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
35 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.

It says stremio is already the newest versio, so do I have it installed now? If so, how do I launch it? It's not appearing in my start menu.

shivasiddharth, 2022-03-17:
Just type `stremio` in the terminal and press enter.

MiroMatton, 2022-03-19:
> When I do this, I get an error that says: Reading package lists... Done Building dependency tree Reading state information... Done Note, selecting 'libfdk-aac1' instead of [...]

I get the same error, found a solution?

jpants36, 2022-03-19:
> I get the same error, found a solution?

Nope, I just gave up. Gone back to using raspicast to watch movies now

MiroMatton, 2022-03-21:
> Nope, I just gave up. Gone back to using raspicast to watch movies now

Can you watch stremio with that ?

shivasiddharth, 2022-03-22:
@MiroMatton Share the output of your:
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue   
sudo nano /etc/apt/sources.list
```

jpants36, 2022-03-22:
> Can you watch stremio with that ?

No, you use apps like cinemahd and BeeTV with it.

shivasiddharth, 2022-03-23:
Run the installation step and share the terminal output !

MiroMatton, 2022-03-23:
[image]

shivasiddharth, 2022-03-24:
Give me the output of `uname -a`

MiroMatton, 2022-03-24:
Linux raspberrypi 5.10.103-v7l+ #1530 SMP Tue Mar 8 13:05:01 GMT 2022 armv7l GNU/Linux

shivasiddharth, 2022-03-24:
Do:
`sudo apt-get update`
and share what it says.

MiroMatton, 2022-03-24:
Get:1 http://deb.debian.org/debian bullseye InRelease [116 kB]
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease
Get:4 http://ftp.us.debian.org/debian bullseye InRelease [116 kB]
Err:1 http://deb.debian.org/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 605C66F00D6C9793
Err:4 http://ftp.us.debian.org/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 605C66F00D6C9793
Reading package lists... Done
W: GPG error: http://deb.debian.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 605C66F00D6C9793
E: The repository 'http://deb.debian.org/debian bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: http://ftp.us.debian.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 605C66F00D6C9793
E: The repository 'http://ftp.us.debian.org/debian bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

shivasiddharth, 2022-03-24:
@MiroMatton Did you go through the notes https://github.com/shivasiddharth/Stremio-RaspberryPi#note ?

MiroMatton, 2022-03-24:
That did it, thx you.

closed 2022-03-24 by shivasiddharth