shivasiddharth/Stremio-RaspberryPi #36: Hardware acceleration on Pi 3?

dialmove, 2022-03-24:
I installed this version on a Raspberry Pi 3 with the latest Raspberry Pi OS, and it runs, but hardware acceleration is disabled so it plays videos at just a couple frames per second.

Is there a way to compile it with hardware acceleration enabled on the Pi 3?

shivasiddharth, 2022-03-25:
Pi 3 is not great for running Stremio. If you want to try the previous build which had ffmpeg, try this https://github.com/shivasiddharth/Stremio-RaspberryPi/tree/4cfccbdce9a70eff3cf36a398c0e3d6046e11546 Download the Repo as a zip, rename the folder and start from https://github.com/shivasiddharth/Stremio-RaspberryPi/tree/4cfccbdce9a70eff3cf36a398c0e3d6046e11546#make-the-scripts-executable

closed 2022-03-25 by shivasiddharth

dialmove, 2022-03-27:
For those who find this thread and want to try, I managed to install it and make it work at a reasonable framerate for 480 and 720p videos following these steps:

- Install [buster version of Raspberry OS](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-05-28/) (not the latest bullseye, which complains of qt5-default package being removed)
- Add a symbolic link to chromium-browser (installed browser is named just chromium, and it doesn't find it)
- Follow the steps at [How To Enable Hardware Acceleration In Chromium On Raspberry Pi OS](https://www.linuxuprising.com/2021/04/how-to-enable-hardware-acceleration-in.html) or at [Smooth YouTube 1080p in Chromium (Raspberry Pi 3 Model B+)](https://www.bodrix.net/2020/01/smooth-youtube-1080p-in-chromium.html)