shivasiddharth/Stremio-RaspberryPi #37: Installation problem - unmet dependencies

JamieBoosh, 2022-04-26:
Hi I have previously been able to install and use Stremio using your installer on my raspberry pi 400. Following a fresh install it no longer works I get the following when running the install step:

> pi-128@raspberrypi:~ $ cd /home/pi-128/Stremio-4.4.142-armhf-32-bit
pi-128@raspberrypi:~/Stremio-4.4.142-armhf-32-bit $ sudo apt-get install ./libfdk-aac1_0.1.6-1_armhf.deb ./stremio_4.4.142-1_armhf.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
libfdk-aac1 is already the newest version (0.1.6-1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
>The following packages have unmet dependencies:
 libqt5webengine5 : Depends: libqt5webengine-data (= 5.11.3+dfsg-2+deb10u1) but 5.15.2+dfsg-3 is to be installed
                    Depends: qtbase-abi-5-11-3
 libqt5webenginecore5 : Depends: libqt5webengine-data (= 5.11.3+dfsg-2+deb10u1) but 5.15.2+dfsg-3 is to be installed
                        Depends: qtbase-abi-5-11-3
 libqt5webenginewidgets5 : Depends: libqt5webengine-data (= 5.11.3+dfsg-2+deb10u1) but 5.15.2+dfsg-3 is to be installed
 libqt5webview5 : Depends: qtbase-abi-5-11-3
                  Depends: qtdeclarative-abi-5-11-2
E: Unable to correct problems, you have held broken packages.

As requested in previous issue threads I have the following outputs:

**cat /proc/cpuinfo:**

>processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

>processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

>processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

>processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

>Hardware	: BCM2711
Revision	: c03130
Serial		: 100000007b55d541
Model		: Raspberry Pi 400 Rev 1.0

**cat /etc/os-release:**
>PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

**cat /etc/rpi-issue:**
>Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage4

**sudo nano /etc/apt/sources.list:**
>deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
\# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
deb http://ftp.us.debian.org/debian/ buster main contrib non-free
deb http://deb.debian.org/debian buster main contrib non-free

shivasiddharth, 2022-04-27:
You are running bullseye 

> VERSION="11 (bullseye)"
> VERSION_CODENAME=bullseye

 but have added buster sources

> deb http://ftp.us.debian.org/debian/ buster main contrib non-free
> deb http://deb.debian.org/debian buster main contrib non-free  

Please check point number 3. I have clearly given two different commands for Bullseye and Buster

closed 2022-04-27 by shivasiddharth

JamieBoosh, 2022-04-27:
What an idiot! Sorry for wasting your time. I will sort that out later when I get home.