shivasiddharth/Stremio-RaspberryPi #38: Can't find the debs

vliapis, 2022-06-05:
Hello, i can't find the debs any longer.

shivasiddharth, 2022-06-05:
I don't understand ! What's the issue ?

vliapis, 2022-06-05:
hi, while it doesnt display on my rpi crhomium, i found it on my x86 machine as per usual, dont know why. Good morning and thank you!

I had an issue where the stremio would exit with a opengl state > 0 or something as soon as the video was buffered and was about to display it (might be able to find the error and post it) and i end up rolling back to a fresh OS (reset the OS) on rpi.

shivasiddharth, 2022-06-05:
Try if you can capture the error next time. Previously the stremio was running on a browser and making Pi slow. These binaries are much better than the previous methods.

closed 2022-12-08 by shivasiddharth