shivasiddharth/Stremio-RaspberryPi #42: Unmet dependencies

souljacker, 2022-09-03:
Hi, thanks for the work and the helpful assistance. 

I`m having issues installing it, though.
Here`s some info

```
uname -a
Linux raspberrypi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux
```

```
sudo apt-get install ./libfdk-aac1_0.1.6-1_arm64.deb  ./stremio_4.4.142-1_arm64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfdk-aac1:arm64' instead of './libfdk-aac1_0.1.6-1_arm64.deb'
Note, selecting 'stremio:arm64' instead of './stremio_4.4.142-1_arm64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libfdk-aac1:arm64 : Depends: libc6:arm64 (>= 2.17) but it is not installable
 stremio:arm64 : Depends: libmpv1:arm64 but it is not installable
                 Depends: qml-module-qt-labs-platform:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtquick-controls:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtquick-dialogs:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtwebchannel:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtwebengine:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qt-labs-folderlistmodel:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qt-labs-settings:arm64 (>= 5.9.5) but it is not installable
                 Depends: librubberband2:arm64 but it is not installable
                 Depends: libuchardet0:arm64 but it is not installable
                 Depends: libqt5webview5-dev:arm64 but it is not installable
                 Depends: libkf5webengineviewer-dev:arm64 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Settordici, 2022-09-19:
I had the same issue because I didn't realize I was using the 64 bit version on a 32 bit system, maybe it could be the same for you?

souljacker, 2022-09-19:
It might be that, yes.

I'll try it out and report back. Thanks!

souljacker, 2022-09-24:
Tried with 32 version, still seems to be the same issue :(

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
 
The following packages have unmet dependencies:
 libqt5webengine5 : Depends: libqt5webengine-data (= 5.11.3+dfsg-2+deb10u1) but 5.15.2+dfsg-3 is to be installed
                    Depends: qtbase-abi-5-11-3
 libqt5webenginecore5 : Depends: libqt5webengine-data (= 5.11.3+dfsg-2+deb10u1) but 5.15.2+dfsg-3 is to be installed
                        Depends: qtbase-abi-5-11-3
 libqt5webenginewidgets5 : Depends: libqt5webengine-data (= 5.11.3+dfsg-2+deb10u1) but 5.15.2+dfsg-3 is to be installed
 libqt5webview5 : Depends: qtbase-abi-5-11-3
                  Depends: qtdeclarative-abi-5-11-2
E: Unable to correct problems, you have held broken packages.
```

Settordici, 2022-09-25:
well, previously it said that the required packages were not installable. With the 32-bit version, it says that dependencies are to be installed. Maybe try installing the required packages? Like libqt5webengine-data.

filipsuk, 2023-04-24:
I have the same error

`Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux`

```
sudo apt-get install ./libfdk-aac1_0.1.6-1_arm64.deb ./stremio_4.4.142-1_arm64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfdk-aac1:arm64' instead of './libfdk-aac1_0.1.6-1_arm64.deb'
Note, selecting 'stremio:arm64' instead of './stremio_4.4.142-1_arm64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libfdk-aac1:arm64 : Depends: libc6:arm64 (>= 2.17) but it is not installable
 stremio:arm64 : Depends: libmpv1:arm64 but it is not installable
                 Depends: qml-module-qt-labs-platform:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtquick-controls:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtquick-dialogs:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtwebchannel:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qtwebengine:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qt-labs-folderlistmodel:arm64 (>= 5.9.5) but it is not installable
                 Depends: qml-module-qt-labs-settings:arm64 (>= 5.9.5) but it is not installable
                 Depends: librubberband2:arm64 but it is not installable
                 Depends: libuchardet0:arm64 but it is not installable
                 Depends: libqt5webview5-dev:arm64 but it is not installable
                 Depends: libkf5webengineviewer-dev:arm64 but it is not installable
E: Unable to correct problems, you have held broken packages.

```

@shivasiddharth any ideas?

shivasiddharth, 2023-04-24:
Most likely its due to not running `sudo apt-get update` after adding the sources. The 64 bit is working without any issue. Please check #44