shivasiddharth/Stremio-RaspberryPi #43: Raspberry Pi OS 64 bit installed from Raspberry Pi Imager reduce the steps needed

Luciano2018, 2022-12-08:
On Raspberry Pi OS 64 bit installed from Raspberry Pi Imager the Steps are just:

**Abrimos la terminal e ingresamos--Get Stremio for Pi OS 64bit opening a Terminal:**
```
cd
wget https://github.com/shivasiddharth/Stremio-RaspberryPi/releases/download/4.4.142/Stremio-4.4.142-arm64-64-bit.zip
```
**Descomprimimos y nos movemos al nuevo directorio--Unzip and move to the new dirctory:**
```
unzip Stremio*.zip
cd Stremio-4.4.142-arm64-64-bit
```
**Instalamos Stremio--Install Stremio:**
`sudo apt-get install -y ./libfdk*.deb ./stremio*.deb -f`

Thanks shivasiddharth

The only issue is in 4k tv with 4k desktop resolution the stream at 2160p is imposible.

closed 2022-12-08 by Luciano2018

shivasiddharth, 2022-12-08:
Thanks for notifying.

a-sarfer, 2023-12-01:
I tried this method on my RPI running on Raspberry Pi OS 64 bit and it doesn't work. There's a dependency issue with libmp1 but it says it's not installable. Any idea on how can I solve this?

Luciano2018, 2023-12-01:
Because you are making the steps on Pi OS Bookworm, the Issue is from December 2022 when the last Pi OS was Bullseye, now libmpv1 is libmpv2 and is another package called libmpv-dev, im trying to test if that fix it, if not we need the experience of @shivasiddharth to make Stremio compatible with Bookworm or follow his steps for compile.

`lu@zero:~/Stremio-4.4.142-arm64-64-bit $ apt search libmpv
Sorting... Done
Full Text Search... Done
celluloid/stable 0.20-2+b1 arm64
  simple GTK+ frontend for mpv

haruna/stable 0.10.3-1 arm64
  Video player built with Qt/QML on top of libmpv

libmpv-dev/stable,now 0.35.1-4 arm64 [installed]
  video player based on MPlayer/mplayer2 (client library dev files)

libmpv2/stable,now 0.35.1-4 arm64 [installed,automatic]
  video player based on MPlayer/mplayer2 (client library)

lu@zero:~/Stremio-4.4.142-arm64-64-bit $ sudo apt install -y ./lib*.deb ./stremio*.deb -f
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_arm64.deb'
Note, selecting 'stremio' instead of './stremio_4.4.142-1_arm64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 stremio : Depends: libmpv1 but it is not installable
E: Unable to correct problems, you have held broken packages.
lu@zero:~/Stremio-4.4.142-arm64-64-bit $`

I've installed:
`sudo apt install libmpv-dev`
but don't make the solve.
Yes, we need to wait experience update to made compatible with Bookworm.

So for the rest of the world, the steps only works on Pi OS 64 Bullseye, apologies.

a-sarfer, 2023-12-01:
Thank you! I'll wait for a Bookworm version then 🙂

shivasiddharth, 2023-12-13:
An update is long overdue. Will work to get new releases.