shivasiddharth/Stremio-RaspberryPi #44: Unmet Dependencies - january 2023

AMassiveBee, 2023-01-21:
I have a RP400, bullseye. Brand new if that makes a difference.
I downloaded the zip file and unzipped it in /home/${USER}/. I then downloaded the bullseye sources, 32bits, and went into the directory.
I tried to perform the installation using the 32bits line of code, but it said I have unmet dependencies.

```
pi@raspberrypi:~ $ cd /home/${USER}/Stremio-4.4.142-armhf-32-bit/  
pi@raspberrypi:~/Stremio-4.4.142-armhf-32-bit $ sudo apt-get install ./libfdk-aac1_0.1.6-1_armhf.deb ./stremio_4.4.142-1_armhf.deb -f   
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 stremio : Depends: qml-module-qtwebengine (>= 5.9.5) but it is not installable
           Depends: libqt5webview5-dev but it is not installable
           Depends: libkf5webengineviewer-dev but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Unsure where to go from here.

shivasiddharth, 2023-01-21:
Give me the output of:
`cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue`

shivasiddharth, 2023-01-21:
And also:
`/etc/apt/sources.list  `

AMassiveBee, 2023-01-21:
```
cat /etc/os-release  
cat /etc/rpi-issue
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: c03131
Serial		: 100000002c202b67
Model		: Raspberry Pi 400 Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f01430c9d8f67a4b9719cc00e74a2079d3834d5d, stage5

```

```
pi@raspberrypi:~/Stremio-4.4.142-armhf-32-bit $ cat /etc/apt/sources.list 
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
deb http://deb.debian.org/debian bullseye main contrib non-free
deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
deb http://deb.debian.org/debian bullseye main contrib non-free
deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
deb http://deb.debian.org/debian bullseye main contrib non-free
deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
deb http://deb.debian.org/debian bullseye main contrib non-free
deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
deb http://deb.debian.org/debian bullseye main contrib non-free
deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
deb http://deb.debian.org/debian bullseye main contrib non-free
deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
deb http://deb.debian.org/debian bullseye main contrib non-free

```

shivasiddharth, 2023-01-21:
GIve me the output of `uname -m`

AMassiveBee, 2023-01-21:
`armv7l`

shivasiddharth, 2023-01-21:
Which version of the Raspberry Pi OS you have ? Lite, Minimal Desktop or Full desktop ? Did you run sudo apt-get update after adding the sources ?

shivasiddharth, 2023-01-21:
After

AMassiveBee, 2023-01-21:
How do i check the version? i only ran the code that's in your instructions.

AMassiveBee, 2023-01-21:
Didn't seem to work.

```
pi@raspberrypi:~ $ cd /home/${USER}/Stremio-4.4.142-armhf-32-bit/  
pi@raspberrypi:~/Stremio-4.4.142-armhf-32-bit $ sudo apt-get install -y ./libfdk*.deb ./stremio*.deb -f
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 stremio : Depends: qml-module-qtwebengine (>= 5.9.5) but it is not installable
           Depends: libqt5webview5-dev but it is not installable
           Depends: libkf5webengineviewer-dev but it is not installable
E: Unable to correct problems, you have held broken packages.

```

shivasiddharth, 2023-01-21:
What image did you select to be written on the SD card ?

AMassiveBee, 2023-01-21:
Probably minimal desktop.

benAkehurst, 2023-01-31:
I've also got the same error showing like @AMassiveBee, any update on how to get around the error? I've tried installing the qml-module-qtwebengine package, but that also didn't install properly. I'm a bit new to the world of Linux

imsilentval, 2023-02-05:
Hi there! I`m the one who posted on reddit, I`ll add in here all the info you asked.

First, I have tried  <sudo apt-get install -y ./libfdk*.deb ./stremio*.deb -f>, this is what I got

        Reading package lists... Done
        Building dependency tree... Done
        Reading state information... Done
        Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
        Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
        Some packages could not be installed. This may mean that you have
        requested an impossible situation or if you are using the unstable
        distribution that some required packages have not yet been created
        or been moved out of Incoming.
        The following information may help to resolve the situation:
        
        The following packages have unmet dependencies:
         stremio : Depends: qml-module-qtwebengine (>= 5.9.5) but it is not installable
                   Depends: libqt5webview5-dev but it is not installable
                   Depends: libkf5webengineviewer-dev but it is not installable
        E: Unable to correct problems, you have held broken packages.

The image was created with the Raspberry Pi Imager v1.7.3, and I chose the first option 
          [image]

Result from cat /proc/cpuinfo

            processor	: 0
            model name	: ARMv7 Processor rev 4 (v7l)
            BogoMIPS	: 38.40
            Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
            CPU implementer	: 0x41
            CPU architecture: 7
            CPU variant	: 0x0
            CPU part	: 0xd03
            CPU revision	: 4
            
            processor	: 1
            model name	: ARMv7 Processor rev 4 (v7l)
            BogoMIPS	: 38.40
            Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
            CPU implementer	: 0x41
            CPU architecture: 7
            CPU variant	: 0x0
            CPU part	: 0xd03
            CPU revision	: 4
            
            processor	: 2
            model name	: ARMv7 Processor rev 4 (v7l)
            BogoMIPS	: 38.40
            Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
            CPU implementer	: 0x41
            CPU architecture: 7
            CPU variant	: 0x0
            CPU part	: 0xd03
            CPU revision	: 4
            
            processor	: 3
            model name	: ARMv7 Processor rev 4 (v7l)
            BogoMIPS	: 38.40
            Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
            CPU implementer	: 0x41
            CPU architecture: 7
            CPU variant	: 0x0
            CPU part	: 0xd03
            CPU revision	: 4
            
            Hardware	: BCM2835
            Revision	: a22082
            Serial		: 00000000f8be24c0
            Model		: Raspberry Pi 3 Model B Rev 1.2

Output of cat /etc/os-release

          PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
          NAME="Raspbian GNU/Linux"
          VERSION_ID="11"
          VERSION="11 (bullseye)"
          VERSION_CODENAME=bullseye
          ID=raspbian
          ID_LIKE=debian
          HOME_URL="http://www.raspbian.org/"
          SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
          BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Output of cat /etc/rpi-issue

          Raspberry Pi reference 2022-09-22
          Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4

Output of cat /etc/apt/sources.list  

          deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
          # Uncomment line below then 'apt-get update' to enable 'apt-get source'
          #deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
          deb http://ftp.us.debian.org/debian/ buster main contrib non-free
          deb http://deb.debian.org/debian buster main contrib non-free
          deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
          deb http://deb.debian.org/debian bullseye main contrib non-free

Output of uname -m

          armv7l

I guess I got everything. Thanks so much for your help! <3

shivasiddharth, 2023-02-06:
@imsilentval Did you run a `sudo apt-get update`  before starting with the installation ?

imsilentval, 2023-02-06:
> @imsilentval Did you run a `sudo apt-get update` before starting with the installation ?

Not at first, but I did try after and then tried installing again, same errors =(

imsilentval, 2023-02-15:
> @imsilentval Did you run a `sudo apt-get update` before starting with the installation ?

I have news: using a OS 64bit, the installation worked flawlessly, but as soon as I open Stremio the rpi starts to freeze and the movie is played almost like a slideshow... Maybe it's because my rpi is kinda old? If you have any tips about this, I would appreciate.

Thanks for everything =)

shivasiddharth, 2023-02-16:
@imsilentval Pi 3 B is not ideal for media streaming. It does not have the hardware to push out high resolution content.

cristianvia, 2023-02-25:
> Hi there! I am the one who posted on reddit, I will add in here all the info you asked.
> 
> First, I have tried <sudo apt-get install -y ./libfdk*.deb ./stremio*.deb -f>, this is what I got
> 
> ```
>     Reading package lists... Done
>     Building dependency tree... Done
>     Reading state information... Done
>     Note, selecting 'libfdk-aac1' instead of './libfdk-aac1_0.1.6-1_armhf.deb'
>     Note, selecting 'stremio' instead of './stremio_4.4.142-1_armhf.deb'
>     Some packages could not be installed. This may mean that you have
>     requested an impossible situation or if you are using the unstable
>     distribution that some required packages have not yet been created
>     or been moved out of Incoming.
>     The following information may help to resolve the situation:
>     
>     The following packages have unmet dependencies:
>      stremio : Depends: qml-module-qtwebengine (>= 5.9.5) but it is not installable
>                Depends: libqt5webview5-dev but it is not installable
>                Depends: libkf5webengineviewer-dev but it is not installable
>     E: Unable to correct problems, you have held broken packages.
> ```
> 
> The image was created with the Raspberry Pi Imager v1.7.3, and I chose the first option [image]
> 
> Result from cat /proc/cpuinfo
> 
> ```
>         processor	: 0
>         model name	: ARMv7 Processor rev 4 (v7l)
>         BogoMIPS	: 38.40
>         Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
>         CPU implementer	: 0x41
>         CPU architecture: 7
>         CPU variant	: 0x0
>         CPU part	: 0xd03
>         CPU revision	: 4
>         
>         processor	: 1
>         model name	: ARMv7 Processor rev 4 (v7l)
>         BogoMIPS	: 38.40
>         Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
>         CPU implementer	: 0x41
>         CPU architecture: 7
>         CPU variant	: 0x0
>         CPU part	: 0xd03
>         CPU revision	: 4
>         
>         processor	: 2
>         model name	: ARMv7 Processor rev 4 (v7l)
>         BogoMIPS	: 38.40
>         Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
>         CPU implementer	: 0x41
>         CPU architecture: 7
>         CPU variant	: 0x0
>         CPU part	: 0xd03
>         CPU revision	: 4
>         
>         processor	: 3
>         model name	: ARMv7 Processor rev 4 (v7l)
>         BogoMIPS	: 38.40
>         Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
>         CPU implementer	: 0x41
>         CPU architecture: 7
>         CPU variant	: 0x0
>         CPU part	: 0xd03
>         CPU revision	: 4
>         
>         Hardware	: BCM2835
>         Revision	: a22082
>         Serial		: 00000000f8be24c0
>         Model		: Raspberry Pi 3 Model B Rev 1.2
> ```
> 
> Output of cat /etc/os-release
> 
> ```
>       PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
>       NAME="Raspbian GNU/Linux"
>       VERSION_ID="11"
>       VERSION="11 (bullseye)"
>       VERSION_CODENAME=bullseye
>       ID=raspbian
>       ID_LIKE=debian
>       HOME_URL="http://www.raspbian.org/"
>       SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
>       BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> ```
> 
> Output of cat /etc/rpi-issue
> 
> ```
>       Raspberry Pi reference 2022-09-22
>       Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4
> ```
> 
> Output of cat /etc/apt/sources.list
> 
> ```
>       deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
>       # Uncomment line below then 'apt-get update' to enable 'apt-get source'
>       #deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
>       deb http://ftp.us.debian.org/debian/ buster main contrib non-free
>       deb http://deb.debian.org/debian buster main contrib non-free
>       deb http://ftp.us.debian.org/debian/ bullseye main contrib non-free
>       deb http://deb.debian.org/debian bullseye main contrib non-free
> ```
> 
> Output of uname -m
> 
> ```
>       armv7l
> ```
> 
> I guess I got everything. Thanks so much for your help! <3

Same exact issue, brand new installed sd 32 bits. Any ideas?

shivasiddharth, 2023-02-26:
I have to get my hands dirty to see what's happening and unfortunately, I never got time. Will try to replicate the issue in coming days.  r/raspberry_pi/ reddit have blocked/banned me, not sure why. So I cannot reply to comments there.

imsilentval, 2023-02-27:
@cristianvia Hi there! Try using the 64 bits version, worked for me. The only problem is that my rpi3 couldn't handle the video, unfortunately =(

[image]
[image]

cristianvia, 2023-02-27:
Hello!
I have reinstalled the 64 bit version as you show in the screenshot.
I can confirm now it works, dude, I'm going to buy you a couple coffees, thanks for your work!

shivasiddharth, 2023-02-27:
@cristianvia Thanks. I really appreciate.

shivasiddharth, 2023-02-27:
I will leave this thread open until I see what's wrong with 32bit installations.

benAkehurst, 2023-03-02:
@shivasiddharth I've tested the 64-bit install and it works. Thanks @cristianvia for the idea!

Kabouik, 2023-07-15:
Are there any plans on making this work on non-Raspbian systems? I use Droidian on my aarch64 phone, which has `apt`, and would love to install Stremio on it natively without using Waydroid and the apk. Unfortunately, I'm encountering the same dependency issues, and official Stremio or Stremio-service builds still don't offer ARM builds.